CytronTechnologies/CytronMotorDriver #3: MD20A not running sample code

DanMBugsy, 2020-04-10:
I have installed the CytronMotorDriver library.  I have connected the Arduino Uno to the MD20A motor driver and to the motor and 12V power supply per the instruction.  When I press either of the test buttons on the board, the motor does turn.  When I try to run the PWM_DIR sketch, it uploads without any errors, but nothing happens.  I test the voltage at the GND and PWM terminals on the MD20A and I read the voltage is changing as per the sketch.

I am at a loss.  Could it be a bad MD20A board?

waiweng83, 2020-04-10:
Yes, it might be a bad board. To verify this, power up the MD20A, use a jumper cable and connect the PWM terminal to the VB+ terminal. Does the motor run?

DanMBugsy, 2020-04-10:
It does NOT run with the PWM terminal connected to the VB+ terminal.

Dan Malone
danmbugsy@gmail.com

DanMBugsy, 2020-04-10:
I got it to work.  The lead I was using for the PWM was bad.

Thanks for your help.

Dan Malone
danmbugsy@gmail.com

waiweng83, 2020-04-13:
It's good to hear that you have sorted it out.
Thanks and have a nice day.

closed 2020-04-13 by waiweng83