CytronTechnologies/CytronMotorDriver #8: Cytron Shield-MDD10

Mr-Caldwell, 2022-12-08:
Hello,

I am having trouble using the sample code for dual motors on an Arduino Uno with a Shield-MDD10. 

It will function normally if I use "setSpeed(255);" or "setSpeed(-255);", but if I set the speed to anything less than 255, ("setSpeed(128)" for example) only one motor will actually run, and the other motor stops moving entirely.

Any ideas on what I could be doing wrong?

waiweng83, 2022-12-19:
Are the PWM & DIR pins defined correctly according to the jumper settings on the SHIELD-MDD10?

Mr-Caldwell, 2023-05-02:
Yes, the jumpers are connected as follow:
PWM1 : pin 9
DIR1 : pin 8
PWM2 : pin 11
DIR2 : pin 13

And that is also how I have them initialized in my code.

waiweng83, 2023-05-03:
Please contact support@cytron.io in that case.
Do share with us the code you used to test it too.

Thanks.

closed 2023-05-25 by waiweng83