CytronTechnologies/CytronMotorDriver #15: Running the motors on MD10 R2 restarts the arduino UNO

vishalchaudhary, 2025-04-24:
Hi I have been using the sketch from Simhub to build a windsimulator using the md10 R2. 

1. I am using 12V powersupply for MD10 R2
2. Arduino UNO is connected to the PC using the USB cable
3. Test button works perfectly without any issue and the fan turns without restarting the UNO
4. Only when the fan is triggered using the UNO it restarts the UNO 

How do I solve this

waiweng83, 2025-04-25:
Can you try it out without the motor connected?

vishalchaudhary, 2025-04-25:
I will try it today and get back, thanks for the prompt response, really appreciate it.

vishalchaudhary, 2025-04-25:
@waiweng83 I just checked, without the fan attached there is no issue of restart, it works perfectly and never restarts the arduino.

waiweng83, 2025-04-28:
Can you share the connection diagram?

vishalchaudhary, 2025-05-03:
@waiweng83 Thanks for the support, seems like the board was faulty, I changed the board and everything is working fine now. I have one question though, I felt the cytron board runs pretty hot after 30mins of use at around 3amps, does it need active cooling ?

vishalchaudhary, 2025-05-04:
Today the cytron board cut power because of heating, can you give me some solution, what kind of active cooling is required for this board